banksy-git/lidl-gateway-freedom #2: root passwd doesn't work

dajrekt, 2021-02-06:
Hi there

I follow the procedure to get the root passwd however the passwd I got doesn't work.
Not sure if I missed something . 

I did as follow:
1. connected board via TTL serial and run putty terminal
2. interrupted boot process and got <RealTek> prompt 
3. read  KEK and AUSKEY
4. run the script and received auskey and root pass
5. power cycled the board with the network attached to boot in normal mode
6. via putty tried to log in with the passwd from the script but unfortunately system says login incorrect

SO I got stuck here for now. 

Can you help?

Ordspilleren, 2021-02-06:
When I ran the script, the output looked something like this (where AUSKEY and PASSWORD are the actual values generated by the script):
```
Auskey: b'AUSKEY'
Root password: b'PASSWORD'
```
The actual Auskey and password are inside the single quotes. So try discarding `b''` and see if that works for you.

dajrekt, 2021-02-07:
Thanks!!! I wouldn't guess that b' is not part of the passwd! You saved me a lot of time!

I have it now working and its great! I love it .

One question .
I copied serialgateway, changed startup and  like I said it is working great, but I still connects to SSH via 2333 not 22. It doesn't bother me much, just would like to know where I misunderstood the process

Thanks again, appreciate it!

banksy-git, 2021-02-07:
> I still connects to SSH via 2333 not 22. It doesn't bother me much, just would like to know where I misunderstood the process

Someone else mentioned that. I might have gone further with the customisation on mine and forgot about it! I will check later today.

closed 2021-02-07 by banksy-git