banksy-git/lidl-gateway-freedom #3: Gateway not starting?

bvlet, 2021-02-07:
I found your website and followed everything to the letter. I managed to get the password, login, put serialgateway in the /tuya/ dir and execute the script. I get a new tuya_start.original.sh, and anew tuya_start.sh. When I use cat to check the new files I think the output is correct:

[](url)[image]

But: After reboot I can't reach the unit through home assistant, and i am still able to SSH to port 2333. SO something is not completely working.
Do you have a hint for me to take me to the next and last step?

Thanks!

MattWestb, 2021-02-08:
I was installing the MobaXterm and is getting the same error so i think its some security things that is not installed / supported in the tuya Linux but its working.
Your startup command looks OK but have you setting the rights OK ?
```
     ┌────────────────────────────────────────────────────────────────────┐
     │                        • MobaXterm 20.6 •                          │
     │            (SSH client, X-server and networking tools)             │
     │                                                                    │
     │ ➤ SSH session to root@192.168.2.60                                 │
     │   • SSH compression : ✔                                            │
     │   • SSH-browser     : ✔                                            │
     │   • X11-forwarding  : ✔  (remote display is forwarded through SSH) │
     │   • DISPLAY         : ✔  (automatically set on remote server)      │
     │                                                                    │
     │ ➤ For more info, ctrl+click on help or visit our website           │
     └────────────────────────────────────────────────────────────────────┘

sh: /usr/bin/xauth: not found
# ls -la /tuya/tuya_start.sh
-rwxrwxr-x    1 1000     1000           32 Feb  1  2021 /tuya/tuya_start.sh
# ls -la /tuya/serialgateway
-rwxr-xr-x    1 root     0           48786 Feb  1  2021 /tuya/serialgateway
#

```

MattWestb, 2021-02-08:
Sorry i was forgetting.
xauth is used by X-server that is not  installed on the tuya GW = OK the terminal cant starting it.

banksy-git, 2021-02-08:
Does "ps" show the gateway is running?

MattWestb, 2021-02-08:
My is always being 127 :
```
  127 root       188 S    /tuya/serialgateway
```
And its looks the 2333 is not being changed of the new startup script (and not patched other things so much).

banksy-git, 2021-02-08:
Ok, I remember - I didn't like the tuyadropbear or its anti brute force mechanism so I disabled it and that caused it to revert to the "standard" dropbear. You can do it too with:

``
if [ ! -f /tuya/ssh_monitor.original.sh ]; then cp /tuya/ssh_monitor.sh /tuya/ssh_monitor.original.sh; fi
echo "#!/bin/sh" >ssh_monitor.sh
``

I'll update the guide accordingly. :)

closed 2021-02-08 by banksy-git

bvlet, 2021-02-08:
Almost works, the echo has to be:

`# echo "#!/bin/sh" >/tuya/ssh_monitor.sh
`
Otherwise you'll get 

`# echo "#!/bin/sh" >ssh_monitor.sh
-sh: can't create ssh_monitor.sh: Read-only file system
`

After that you can SSH through port 22

banksy-git, 2021-02-08:
Fixed! Thanks @bvlet