banksy-git/lidl-gateway-freedom #4: Error in prebuilt serialgateway?

bvlet, 2021-02-08:
After not being able to connect to port 8888 after a reboot, I tried to start serialgateway manually:

`
# /bin/sh /tuya/serialgateway
/tuya/serialgateway: line 1: syntax error: "(" unexpected
#
`
Could it be that there's an error in the provided prebuilt serialgateway.bin?

banksy-git, 2021-02-08:
It's not a shellscript! Run without /bin/sh :)

banksy-git, 2021-02-08:
Also, what's in the ps list after you reboot?

bvlet, 2021-02-08:
`

sh: /usr/bin/xauth: not found
# ps
  PID USER       VSZ STAT COMMAND
    1 root      1292 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW<  [ksoftirqd/0]
    4 root         0 RW   [kworker/0:0]
    5 root         0 SW<  [kworker/0:0H]
    6 root         0 SW   [kworker/u2:0]
    7 root         0 SW<  [khelper]
    8 root         0 SW<  [writeback]
    9 root         0 SW<  [bioset]
   10 root         0 SW<  [crypto]
   11 root         0 SW<  [kblockd]
   12 root         0 SW   [kswapd0]
   13 root         0 SW   [kworker/0:1]
   14 root         0 SW   [fsnotify_mark]
   20 root         0 SW   [mtdblock0]
   21 root         0 SW   [mtdblock1]
   22 root         0 SW   [mtdblock2]
   23 root         0 SW   [mtdblock3]
   24 root         0 SW   [mtdblock4]
   25 root         0 SW<  [deferwq]
   26 root         0 SW   [kworker/u2:1]
   27 root         0 SW<  [kworker/0:1H]
   96 root         0 SWN  [jffs2_gcd_mtd4]
  118 root      1552 S    dropbear -P /var/run/dropbear.1.pid -p 22 -K 300
  120 root      1292 S    init
  125 root      1288 S    udhcpc -i eth1 -s /bin/udhcpc.script -p /var/run/udhc
  126 root      1872 R    dropbear -P /var/run/dropbear.1.pid -p 22 -K 300
  127 root      1292 S    -sh
  130 root      1288 R    ps
#
#
`

banksy-git, 2021-02-08:
Did the serial gateway make it across ok? Check file size matches.
What happens when you run just /tuya/serialgateway?

bvlet, 2021-02-08:
Looks like it....

[image]

In my explorer it's 47,6 kB.

I also tried the commands over Serial connection, logged in as root, but that makes no difference...

Running just /tuya/serialgateway gives an error:

```
# /tuya/serialgateway
Illegal instruction
#
```

banksy-git, 2021-02-09:
Can you run this on your device. Assuming you got serialgateway.bin from [here](https://paulbanks.org/download/files/lidl-zigbee/serialgateway.bin) It should be the same as this:
```
# md5sum /tuya/serialgateway 
9f783720da1392f2f6589c2debb91bf0  /tuya/serialgateway
```
If it matches, can you post output of:
```
cat /proc/cpuinfo
```

bvlet, 2021-02-09:
Nope, doesn't match....
Ik re-downloaded with your link and replaced the file. output is this:

```
sh: /usr/bin/xauth: not found
# md5sum /tuya/serialgateway
e1602a45d7033fed5372ee93a9a2599d  /tuya/serialgateway
#

```

CPUinfo is like this:

```
# cat /proc/cpuinfo
system type             : RTL8196E
machine                 : Unknown
processor               : 0
cpu model               : 52481
BogoMIPS                : 398.13
tlb_entries             : 32
mips16 implemented      : yes
#

```

bvlet, 2021-02-09:
WHen I check your file in powershell, I see that I do get the correct MD5hash.....

```
PS C:\> Get-FileHash -Algorithm MD5 serialgateway.bin

Algorithm       Hash                                                                   Path
---------       ----                                                                   ----
MD5             9F783720DA1392F2F6589C2DEBB91BF0                                       C:\serialgateway.bin

PS C:\>
```

So somewhere the file is changing?

bvlet, 2021-02-09:
YAY! I found it! At 3:19 AM....
I'm doing this from my work laptop which is Windows 10. So I used powershell to issue the cat command to transfer the serialgateway.bin. For some reason this "corrupts" the file. 
I just installed WSL(Ubuntu) and issued the command from there. After checking the MD5 tag on the hub and finding that it is identical to the one you sent me, I rebooted and voila! HA immediately connects through port 8888!

Thank you for your support. As always it was a Windows problem at last..... :) Maybe you should put that in your instructions ;)

closed 2021-02-09 by bvlet