banksy-git/lidl-gateway-freedom #8: Root password procedure does not work for Aldi Lightway Zigbee gateway

challs, 2021-04-13:
I tried using this procedure with the Aldi Lightway Smart Home gateway, which appears to be the same as the Lidl variant inside. I was able to connect via serial terminal, read the flash sections and generate a root password. But the password is not accepted on the serial terminal or command line.

I will see if I can use the original method of playing with the squashfs to get control of the device. For now, this issue is just for information in case anyone else is thinking of trying it.

chaisaeng, 2021-05-07:
can you get the root password now? I having the same issue as yours probably. I bought mine from aliexpress the pcb is the same and I can access the boot loader screen via usb-ttl ftdi adapter. the procedure for getting the key all return FFFFFF  like below and the decode script is throwing exception if I input the parameter according to what I get from the device

FLR 80000000 401802 16
Flash read from 00401802 to 80000000 with 00000016 bytes ?
(Y)es , (N)o ? --> y
Flash Read Successed!
DW 80000000 4
80000000: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF
FLR 80000000 402002 32
Flash read from 00402002 to 80000000 with 00000032 bytes ?
(Y)es , (N)o ? --> y
Flash Read Successed!
DW 80000000 8
80000000: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF
80000010: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF

challs, 2021-05-07:
> can you get the root password now? I having the same issue as yours probably.
My flash read came back with other data, not just FFFF. So it's not exactly the same as yours.

However, the tool for recreating the root filesystem should work for you too. There's more information on the forum thread.

https://community.home-assistant.io/t/hacking-the-silvercrest-lidl-tuya-smart-home-gateway/270934/136?u=challs
and
https://community.home-assistant.io/t/hacking-the-silvercrest-lidl-tuya-smart-home-gateway/270934/142?u=challs

chaisaeng, 2021-05-08:
Thanks, I'll look into that links. BTW. I try looking into the flash using FLR on my device at different offset. there are some locations that not FF but I can not identify that it is the kek or aus key needed to decode as it's password so I guess my device may have the kek and aus key stored in different location as the one in this project.

grw1983, 2021-05-18:
I have the 141M100GW Zigbee Gateway from ALDI. root password generation did work for me.

chaisaeng, 2021-05-19:
> I have the 141M100GW Zigbee Gateway from ALDI. root password generation did work for me.

If you did not pair your device with smart life app yet. try th password `tuya123` to see you can get access to root.

challs, 2021-05-22:
> I have the 141M100GW Zigbee Gateway from ALDI. root password generation did work for me.

That's great news. So the procedure is working for at least some gateways, as long as they have not been connected to the internet (since the password will be changed as soon as this happens)

cvictor, 2021-08-11:
> I tried using this procedure with the Aldi Lightway Smart Home gateway, which appears to be the same as the Lidl variant inside. I was able to connect via serial terminal, read the flash sections [...]

For me it didn't work when I copy/pasted the password to the serial console (SerialTools on Mac). Typing it in manually worked. Maybe that helps.

cgringmuth, 2023-06-30:
> can you get the root password now? I having the same issue as yours probably. I bought mine from aliexpress the pcb is the same and I can access the boot loader screen via usb-ttl ftdi adapter. the [...]

I have the same issue. Did you figure out how to solve it?

I also tried to download the image with `dump.py`. But it seems broken. After extracting to `squashfs-root` `/etc/passwd` is not there. It is a symlink to `/tuya/config/passwd`. But `tuya` folder is completely empty.

jonny190, 2023-10-03:
Having the same issue with a rev 1.0.2 board

robsonfelix, 2024-11-18:
same issue here.