banksy-git/lidl-gateway-freedom #12: [Request] More information on Gaining initial access to the device using modified SquashFS

chaisaeng, 2021-05-07:
From the article to Gaining access to the device
Quote
"3) We replaced the /etc/passwd symlink with a passwd file we created with a known root password."
Question: the password file to be put in the unsquashfs filesystem structure, is it need to be format as shardow passwd file or just plain text format like in the old-time linux passwd file. where should it be located in the file system i.e. in root directory and change the symlink to /etc/passwd 
Quote
"6) Finally we TFTP'd the newroot.bin file to the device and used the bootloader command FLW to write it to flash."
Can you outline the process to in the Quote (step 6 in the article)  
Currently I have a dump of the device from your dump_flash.py and unsquashfs it into my virtualbox debian vm
I got the directory contains all the files from the SquashFS dumped. as I'm not very familiar with the process of flashing the device with tftp I need more detail on how to do that on my device. 
I got device but can not get the root password fron the process you mentioned in the article so the alternative is to reflash it with modified SquashFS flash image to gain initial access to the device.
I hope this will also help other that fail to get root password of the device as I am.

Thanks very much.

challs, 2021-05-07:
> Question: the password file to be put in the unsquashfs filesystem structure, is it need to be format as shardow passwd file or just plain text format like in the old-time linux passwd file. where should it be located in the file system i.e. in root directory and change the symlink to /etc/passwd

It's shadow format. See [my forum post](https://community.home-assistant.io/t/hacking-the-silvercrest-lidl-tuya-smart-home-gateway/270934/136?u=challs) for more details (look for the `mkpasswd` command.

> Can you outline the process to in the Quote

See [the Upload Image part](https://community.home-assistant.io/t/hacking-the-silvercrest-lidl-tuya-smart-home-gateway/270934/136#upload-the-image)

chaisaeng, 2021-05-08:
> It's shadow format. See [my forum post](https://community.home-assistant.io/t/hacking-the-silvercrest-lidl-tuya-smart-home-gateway/270934/136?u=challs) for more details (look for the `mkpasswd` [...]

Thanks,  I looked into the forum it look like I have found new hope. BTW for  the passwd file you mention it's shadow format which mean in the file if I only having root user then it is only 1 line as followed
root:rvE2CYlmZo43R 
or 2 file with different content below  
file: passwd
root:x:0:0:root:/root:/bin/bash
and
file: shardow
root:rvE2CYlmZo43R

Thanks very much for pointing me to the information I needed.

challs, 2021-05-08:
> for the passwd file you mention it's shadow format which mean in the file if I only having root user then it is only 1 line as followed
> or 2 file with different content below

Sorry, I misread your first question and my answer was confusing.

There is only 1 file ("like in the old-time linux passwd file") There is no shadow password file needed.

chaisaeng, 2021-05-09:
This issue was resolved in the forum mentioned by @challs

closed 2021-05-09 by chaisaeng