banksy-git/lidl-gateway-freedom #13: do not use ssh-copy-id

letroll, 2021-06-05:
first of all, thank you for the article. Just a ticket to warn. I had the bad idea to make an ssh-copy-id ... I'm not going to be able to access ssh anymore :'(

MattWestb, 2021-06-05:
I was playing with using the local comport and was blocking the login with SSH and local console (kernel logging was interference with the local console).
I was flashing back the original partitions and was unbricking the ZBB.

If you have backup of the partitions of your  flash you can do the same.
Look in #6 and other post and you can do it.

letroll, 2021-06-06:
Well, I think I've solved my case...
I didn't understand that I could have root access through the console if I didn't do the boot escape.

Having obtained my root password with the tutorial, I was able to get away with it

closed 2021-06-06 by letroll