banksy-git/lidl-gateway-freedom #15: Impossible to set any other channel exept channel 25.

gribouk, 2021-07-07:
Hello,
I have a problem assigning any channel other than 25 to TYGWZ-01 coordinator.

 I've used this gateway with my old network at channel 25. Now I try to build a new network with it in the neighboring room, while all old devices vere deleted from the ZHA configuration for this coordinator, the configuration was deleted also, and the coordinator is moved to brand new raspberry pi.

Just installed fresh container image for HA and try to add the coordinator to ZHA. But no metter what I try it keep assigning 
channel 25 to it! 

How do I make it run channel 23 then?

My config:
```
zha:
  zigpy_config:
    network:
      channel: 23             # What channel the radio should try to use.
      channels: [21, 22, 23]  # Channel mask
```

MattWestb, 2021-07-07:
ZHA have no (current working) way to changing channel so you need using bellows and doing one `leave`  and you can also doing one `form` but then you need doing little more parameters.

If you have doing one `leave` you can start HA and its shall using the parameters you have putting in the config. 
Look in the HA log for see the parameters its using.

The` channels: [21, 22, 23]` shall being `11, 15, 20, 25` if you is having "normal devices".
Its defining the primary channels that is first used the searching one network and if you is putting not all primary channels devices can having problem (re)joining the system.

gribouk, 2021-07-07:
1. How do I use bellows than? what exactly should I do?

2. The HA log:
```
Node type: EmberNodeType.COORDINATOR, Network parameters: EmberNetworkParameters(extendedPanId=95:fe:4e:27:16:3a:ac:56, panId=0x7a6b, radioTxPower=13, radioChannel=25, j$
2021-07-07 20:42:47 DEBUG (MainThread) [bellows.ezsp.protocol] Send command setPolicy: (<EzspPolicyId.TC_KEY_REQUEST_POLICY: 5>, <EzspDecisionId.ALLOW_TC_KEY_REQUESTS: 81>)
```
3. I cannot use primary channels since these are already occupied with other coordinators - I run 11 coordinators network at home (do not ask me why, it's a long story...). So, the question is if having channel 25 for coordinators in the neighboring rooms is a problem? If so, than I have to moove to secondry channels.

gribouk, 2021-07-07:
4. Are you saying that using other channel than 11, 15, 20, 25 will cause problems?

MattWestb, 2021-07-07:
If using this you shall using  channel 23 and still can find devices on the normal primary zigbee channels.
```
zha:
  zigpy_config:
    network:
      channel: 23             # What channel the radio should try to use.
      channels: [11, 15, 20, 25]  # Channel mask
```
The `channels `is only being used the devices is searching network and the `channel` is the channel your network is using.

Then you is using docker its easiest edit the parameters for your com port so HA cant starting ZHA.
In your config directory `.storage\core.config_entries` find the entries "path".
```
            {
                "entry_id": "ca43d86de6df6304c3dc0c305eda4d44",
                "version": 2,
                "domain": "zha",
                "title": "socket://192.168.2.60:8888 (tuya ZBGW)",
                "data": {
                    "device": {
                        "baudrate": 115200,
                        "path": "socket://192.168.2.60:8888"
                    },
                    "radio_type": "ezsp"
                },
```
(example from my config).
Put one x in the end of the port so its not being OK like this `"path": "socket://192.168.2.60:8888X"`.
Saving and restart HA and its cant starting ZHA but you can using bellows in docker CLI.
Do your bellows commands and then redo the changes you was doing in the first step and restarting HA and ZHA shall starting normal.

gribouk, 2021-07-07:
So, you say:
0. I change zha confing as in your example
1. I go to .storage and add 'x' to "path" field of JASON object
2. I log to container bash and do CLI commands

What are these CLI commands? Like I have to stop HA or what should I do? Will I have to rebuild a network after that?

MattWestb, 2021-07-07:
Its the HA container so you shall have ZHA and all components installed and working.
Start with` bellows -d socket://192.168.2.60:8888 info`(your ip address to the ZBGW need being updated).
For leaving its command `leave `instead of `info`.

gribouk, 2021-07-07:
OK, thabks, I've got that. Looks like I've done 'leave' command.

Should I reboot now with initial HA setup? Or some other CLI commands?

gribouk, 2021-07-07:
Super! Worked! Thanks alot!

MattWestb, 2021-07-07:
No need only redo the config changes and restarting HA and ZHA is starting OK and if no network is formed ZHA shall forming on new with your parameters in the HA config.

You can also forming on network with bellows but i dont remember the parameters but its have hel in the CLI for doing it.

closed 2021-07-07 by gribouk