banksy-git/lidl-gateway-freedom #16: How to recover from passwd

codedmind, 2021-07-13:
Hello, 

After doing all the steps i try to add a new user, but get read only system, then try change the password, i get sucess, but then after reboot, cannot login with the new password or the one the i retrive from the script.

The gateway appear to work normally but cannot login via console or ssh..

Thanks

codedmind, 2021-07-13:
In the page when you refer Gaining initial access, the dump is made with device in RealTek bootloader or in normal boot?

MattWestb, 2021-07-13:
All things made before have getting root access is from the bootloader.

Then you is having the root password you can doing one normal boot and login as root and the best way is with serial console and not SSH if having problems.

codedmind, 2021-07-13:
> All things made before have getting root access is from the bootloader.
> 
> Then you is having the root password you can doing one normal boot and login as root and the best way is with serial console and not SSH if having problems.

I'm not able to login even with console... i try restart the process to get the password, but the password the script returns is the same as the first time... but even with that i cannot login

challs, 2021-07-13:
>  i try restart the process to get the password, but the password the script returns is the same as the first time... but even with that i cannot login

Hi. Did you connect the gateway to the internet before you did the process? People have reported that the root password was changed after connecting to the internet. If that is the case, you will need to edit the squashfs image and upload a modified version.

codedmind, 2021-07-13:
> Hi. Did you connect the gateway to the internet before you did the process? People have reported that the root password was changed after connecting to the internet. If that is the case, you will [...]

I only attach it to the network cable... that network cable have internet... i don't install any app or do any kind of configuration... do the software call to internet alone?

challs, 2021-07-13:
> I only attach it to the network cable... that network cable have internet... i don't install any app or do any kind of configuration... do the software call to internet alone?

Yes, it contacts the tuya servers on startup - that's how it becomes available through the cloud. And that process seems to change the root password. I had this problem myself. There's details of how you can use `rootfs_tool.py` in the forum thread [here](https://community.home-assistant.io/t/hacking-the-silvercrest-lidl-tuya-smart-home-gateway/270934/150)

codedmind, 2021-07-13:
@challs so i should dump the rootfs to get the password?

challs, 2021-07-13:
> @challs so i should dump the rootfs to get the password?

Yes.

> I can't follow the topics in the link you posted... can't find toolfs_tool.py reference

That's rootfs_tool.py, (begins with 'r' not 't')

Look for chaisaeng's post, which begins:

```
Hi challs,
I’m trying to get root access to my device but not quite success yet.
I follow the instruction on [Hacking the Silvercrest (Lidl) Smart Home Gateway - Paul Banks DOT Org ](https://paulbanks.org/projects/lidl-zigbee/) as followed

* Execute: `python dump_flash.py --serial-port /dev/ttyUSB0 --output-file rootfs.bin --start-addr 0x200000 --end-addr 0x400000` ==> This step took me around 40 Minutes
```

codedmind, 2021-07-13:
@challs ok find it.

This step are all in the a host computer, none of them are in the device correct?

challs, 2021-07-13:
Yes that's correct. `dump_flash.py` runs the needed commands through the serial line from the host computer.

codedmind, 2021-07-13:
@challs that i already do, but the next one are the the host also ?

challs, 2021-07-13:
Those that say 'in bootloader screen' (`ipconfig`, `autoburn`, `FLW`) are to be executed on the device.
All the others such as `sudo unsquashfs` need to be executed on the host computer.

codedmind, 2021-07-13:
@challs ok i think i can follow everything... only missing the right way to upload the file ... can you point me the right direction please?

challs, 2021-07-13:
The upload part is
> Upload the file newroot.bin in binary format using tftp

I used this command:

```
$ tftp 172.28.1.6 -m binary -c put newroot.squashfs
```

There's a slightly more verbose description in [my earlier comment](https://community.home-assistant.io/t/hacking-the-silvercrest-lidl-tuya-smart-home-gateway/270934/136?u=challs) in the thread. 

Also, it is important to make sure you use binary mode tftp.

codedmind, 2021-07-13:
@challs well now i'm loose... if i upload the newroot.squashfs where will we use the newroot.bin ?

challs, 2021-07-13:
I did my conversion earlier than chaisaeng, who had an easier workflow when he documented the list. So you should adjust the tftp command to use the name of the file which you have, which is probably newroot.bin.

The tftp command is executed on the host. I'm using linux so it might be slightly different on a Windows machine.

codedmind, 2021-07-13:
@challs i'm also using linux, my tftp server is on 192.168.1.90... what should i do? I'm a little confused about all the differents ips in the posts

codedmind, 2021-07-13:
Ok its is done

On bootload ipconfig 192.168.1.92
On host that is with the ip 192.168.190, but  the tftp command i must use also 192.168.1.92
tftp 192.168.1.90 -m binary -c put newroot.iso

closed 2021-07-13 by codedmind