banksy-git/lidl-gateway-freedom #18: Dual mode - local and Tuya cloud

JFox-sk, 2021-11-17:
Hi,
when I hack the Gateway using these scipts, can I still using Tuya mobile app (aka dual mode: local + tuya cloud)? 
Thank you!

JFox-sk, 2021-11-25:
So, I tried it, and the answer is - no, I can not use both modes at the same time, but I can easily switch between the modes.

closed 2021-11-25 by JFox-sk

MattWestb, 2021-11-25:
I think (= not knowing) its only renaming the start script and copy the original back and its starting tuya app.

Perhaps tuya is updating the box and you is loosing the hacked start script but  the root password shall being OK.

Also it can being that tuya app is not liking the updated Zigbee firmware.

Great that you have testing it !!

JFox-sk, 2021-11-25:
Yes @MattWestb just rename the start script (or change symbolic link).

MattWestb, 2021-11-25:
Did you upgrading the firmware on the Zigbee module ?
And is yes did tuya cloud like it or downgrading it ?
I have not testing if tuya is downgrading the Zigbee module but im curious  if they is having making upgrading then LIDL and more  customer have getting new Zigbee certificate and i have not reversing my ZBGWs for testing (they is in production on my test network and sniffing Zigbee trafic).