banksy-git/lidl-gateway-freedom #20: Content tuya_start.original.sh

burnbabyburn, 2022-02-14:
Hi,

can someone post the content of the original tuya_start.sh?

frtz13, 2022-02-14:
this is my tuya_start.original.sh

```
#!/bin/sh
#Usage: sh tuya_start.sh UserAppRunDir
#=======================================================================
DEFULT_APP_RUN_DIR=/tuya
TY_START_CHILDREN_SHELL=tuya_start_children.sh
def_jsonvalue_NULL="defaultValue"

#echo "Tuya Gateway Application Normal Srart $0 UserAppRunDir:${1} JsonFile Path:${2}"
echo "Tuya Gateway Application Normal Srart $0 UserAppRunDir:${1}"
#set app run dir
app_run_dir=$DEFULT_APP_RUN_DIR

if [ -d "$1" ];then
        app_run_dir=${1%*/}
        echo "set run_dir:${app_run_dir}"
elif [ -n "$TY_ENV_APP_RUN_DIR" ];then
        app_run_dir=$TY_ENV_APP_RUN_DIR
        echo "find old TY_ENV_APP_RUN_DIR:${app_run_dir}"
else
        echo "set defult run_dir:${DEFULT_APP_RUN_DIR}"
fi
export TY_ENV_APP_RUN_DIR=$app_run_dir
LD_LIBRARY_PATH=$app_run_dir:$LD_LIBRARY_PATH
export LD_LIBRARY_PATH
echo "TY_ENV_APP_RUN_DIR=${app_run_dir}"
UserAppRunDir=$app_run_dir
JSON_PARSER_SH=${UserAppRunDir}/json_parser.sh

#load platform configure file
DEFULT_PLATFORM_CFG_FILE=${app_run_dir}/def.cfg
user_cfg_file=$DEFULT_PLATFORM_CFG_FILE
if [ -f "$DEFULT_PLATFORM_CFG_FILE" ];then
        #get user cfg file path
        jsonkey_USER_CFG_FILE="user_cfg_file"
        jsonvalue_USER_CFG_FILE=`sh ${JSON_PARSER_SH} ${DEFULT_PLATFORM_CFG_FILE} ${jsonkey_USER_CFG_FILE} | sed 's/\"//g'`
        if [ "$jsonvalue_USER_CFG_FILE" = "$def_jsonvalue_NULL" ] || [ ! -f "$jsonvalue_USER_CFG_FILE" ] ;then
                echo "get user cfg file error, load defult cfg file"
                user_cfg_file=$DEFULT_PLATFORM_CFG_FILE
        else
                echo "get user cfg file success."
                user_cfg_file=$jsonvalue_USER_CFG_FILE
        fi
else
        echo "defult cfg does not exist."
        exit 0
fi
echo "load platform configure file:${user_cfg_file}"
export TY_ENV_USER_CFG_FILE=$user_cfg_file
#sh $TY_PLATFORM_CFG_PARSER_SHELL $app_run_dir $user_cfg_file
JsonFile=$user_cfg_file

#tmp_dir
jsonkey_USER_TMP_DIR="tmp_dir"
def_jsonvalue_USER_TMP_DIR="/tmp"
jsonvalue_USER_TMP_DIR=`sh ${JSON_PARSER_SH} ${JsonFile} ${jsonkey_USER_TMP_DIR} | sed 's/\"//g'`
if [ "$jsonvalue_USER_TMP_DIR" == "$def_jsonvalue_NULL" ];then
    jsonvalue_USER_TMP_DIR=$def_jsonvalue_USER_TMP_DIR
fi
export TY_ENV_USER_TMP_DIR=${jsonvalue_USER_TMP_DIR%*/}

#platform
jsonkey_PLATFORM="platform"
def_jsonvalue_PLATFORM="RTL8196E"
jsonvalue_PLATFORM=`sh ${JSON_PARSER_SH} ${JsonFile} ${jsonkey_PLATFORM} | sed 's/\"//g'`
if [ "$jsonvalue_PLATFORM" == "$def_jsonvalue_NULL" ];then
    jsonvalue_PLATFORM=$def_jsonvalue_PLATFORM
fi
export TY_ENV_PLATFORM=${jsonvalue_PLATFORM}

#wan_interface
jsonkey_WAN_IF_NAME="wan_interface"
def_jsonvalue_WAN_IF_NAME="eth1"
jsonvalue_WAN_IF_NAME=`sh ${JSON_PARSER_SH} ${JsonFile} ${jsonkey_WAN_IF_NAME} | sed 's/\"//g'`
if [ "$jsonvalue_WAN_IF_NAME" == "$def_jsonvalue_NULL" ];then
    jsonvalue_WAN_IF_NAME=$def_jsonvalue_WAN_IF_NAME
fi
export TY_ENV_WAN_IF_NAME=$jsonvalue_WAN_IF_NAME

app_fold1=${app_run_dir}/tuya_user1
app_fold2=${app_run_dir}/tuya_user2
user_path=${app_fold1}

#restart dhcp
killall -9 udhcpc
echo 4 > /proc/sys/net/ipv4/tcp_syn_retries
killall udhcpc
udhcpc -i ${TY_ENV_WAN_IF_NAME} -s ${app_run_dir}/udhcpc.script -p /var/run/udhcpc0.pid & ##????

#create user_tmp dir
if [ ! -d "$TY_ENV_USER_TMP_DIR" ]; then
        mkdir -p "$TY_ENV_USER_TMP_DIR"
fi

default() {
        echo "Into default funtion"
        user_path=$app_fold1
        if [ ! -d $user_path ];then
                echo "Error: no run dir:${user_path}"
                user_path=$app_fold2
                if [ ! -d $user_path ];then
                        echo "Error: no run dir:${user_path}"
                        exit 0
                else
                        echo "tuya_start_dir=${user_path}" > ${app_run_dir}/start.conf
                fi
        else
                echo "tuya_start_dir=${user_path}" > ${app_run_dir}/start.conf
        fi
}

cd $app_run_dir
if [ ! -r "$app_run_dir" ]; then
    echo "dir:${app_run_dir} error"
        exit -1
fi

if [ ! -w "$app_run_dir" ]; then
    echo "dir:${app_run_dir} read only!"
else
        if [ -s start.conf ];then
                echo "start.conf is exist"
                user_path=`cat start.conf | grep tuya_start_dir | cut -d "=" -f 2`
        else
                echo "start.conf is not exist"
                user_path=/tytest123
        fi

        if [ ! -d $user_path ];then
                echo "$user_path is not exist"
                default
        else
                if [ "$user_path" != "$app_fold1" ] && [ "$user_path" != "$app_fold2" ];then
                        echo "$user_path error."
                        default
                fi
        fi
fi

echo "current run dir:$user_path"
cd $user_path
./$TY_START_CHILDREN_SHELL $app_run_dir $user_cfg_file &

```

burnbabyburn, 2022-02-14:
Thank you very much @frtz13 
I am currently dumping flash to recreate a working rootfs to escape my bootloop. I'll report back to document the recovery steps if one didn't make a backup and close this issue afterwards :)

frtz13, 2022-02-14:
here we go: ssh_monitor.original.sh

```
#!/bin/sh
SSH_FAILED_CNT_FILE=/tuya/ssh/cnt

sleep 3
mkdir -p /tuya/ssh

failed_cnt=0
sleep_time=0

if [ -f "$SSH_FAILED_CNT_FILE" ];then
                failed_cnt=`cat $SSH_FAILED_CNT_FILE`
fi

if [ ! -z "$1" ];then
    test_login_flag=1

        killall tuyadropbear
        sleep 1
        rm -f /var/run/dropbear.tuya.pid

        let failed_cnt=$failed_cnt+1
        echo $failed_cnt > $SSH_FAILED_CNT_FILE

        if [ $failed_cnt -gt 5 ];then
                let sleep_time=432000
        elif [ $failed_cnt -eq 1 ];then
                let sleep_time=90
        elif [ $failed_cnt -eq 2 ];then
                let sleep_time=300
        elif [ $failed_cnt -eq 3 ];then
                let sleep_time=3600
        elif [ $failed_cnt -eq 4 ];then
                let sleep_time=86400
        elif [ $failed_cnt -eq 5 ];then
                let sleep_time=172800
        fi
else
        killall dropbear
        let sleep_time=21600*$failed_cnt
fi

echo "[tuya failed login] $failed_cnt times, sleep $sleep_time secs." >> /tmp/tuya.log
sleep $sleep_time

/tuya/tuyadropbear -P /var/run/dropbear.tuya.pid -p 2333 -K 300 -T 1#

```

burnbabyburn, 2022-02-14:
thank you again :+1: 
i just needed to hardcode the last line in tuya_start.sh to 
`./tyZ3Gw /tuya/tuya_user1/ /tuya/def.cfg &`
get the gateway up and running again

closed 2022-02-14 by burnbabyburn