banksy-git/lidl-gateway-freedom #21: Issue decoding password via script

myhobby2, 2022-02-25:
Thanks for your time and effort in this project! I've 2 MOES branded Zigbee bridges that are internally the same as you use. 

Let's start that I was able to obtain the root password from bridge nummer 1 (in another post I will come back to that one). Following thw same procedure I'm able to get both KEK/AUSKEY, but when re-using the script to decrypt the root password, I run into a problem:

root# python3 lidl_auskey_decode.py
Enter KEK hex string line>5A5AA5A5        401A4000        8F7B238C        001AD582
Encoded aus-key as hex string line 1>13F436F6        E44A50B4        5AF2F409        7AE9D394
Encoded aus-key as hex string line 2>8E335DD1        C67C31D9        63B3EBD5        2C7C790A
Traceback (most recent call last):
  File "lidl_auskey_decode.py", line 64, in <module>
    print("Auskey:", auskey.decode("ascii"))
UnicodeDecodeError: 'ascii' codec can't decode byte 0xaa in position 2: ordinal not in range(128)

Any ideas ?

Ninic0c0, 2022-03-02:
@myhobby2 

You have copy all the line (with 80000000: and spaces) as follow:

python3 lidl_auskey_decode.py
Enter KEK hex string line>80000000:       XXXXXXXX        XXXXXXXX        XXXXXXXX        XXXXXXXX
Encoded aus-key as hex string line 1>80000000:       XXXXXXXX        XXXXXXXX        XXXXXXXX        XXXXXXXX
Encoded aus-key as hex string line 2>80000010:       XXXXXXXX        XXXXXXXX        XXXXXXXX        XXXXXXXX
Auskey: *************
Root password: *************

myhobby2, 2022-03-02:
Thanks for your support. Just succeeded in retrieving the password after some attempts. I had to take out all spaces from the aus-key, although the other box password was done simply by copy/paste (including the spaces)... BTW: I never included the 80000000: or 80000010: part of the lines. 

I'm at least happy that I was able to login and change the passw. Thanks for your efforts in this matter BTW

fjritaine, 2022-10-01:
Hi,
I'm encountering the exact same issue : 
Traceback (most recent call last):
  File "/Users/fritaine/PERSO/lidl_auskey_decode.py", line 64, in <module>
    print("Auskey:", auskey.decode("ascii"))
UnicodeDecodeError: 'ascii' codec can't decode byte 0x84 in position 1: ordinal not in range(128)

I've noticed that my KEK and the first line of my aus-key are indentical, is this normal ? or maybe they did a few software changes since this decoding script has been released ?

Would be glad to contribute and help.
Thanks :)

fjritaine, 2022-10-02:
Hey, I solved my problem and want to share :
- I did a factory reset of the gateway thanks to the Lidl Mobile App (removing the gateway from your app actually wipes all your cloud data from it)
- I changed the serial console I was using which seems to input phantom stuff, thus the commands to read the memory weren't the right ones. I guess that's why my KEK and the first line of my aus-key were the same. I switched to "Serial" from Decisive Tactics on MacOS.

Hope that helps!

muekno, 2024-01-26:
Same problem, new gateway never announced in the cloud.
Did copy all with 80000: ..... including spaces
 same as  at [fjritaine](https://github.com/fjritaine)   I've noticed that my KEK and the first line of my aus-key are indentical, is this normal
tried x times
I am in IT since 40+ years, have good knowledge in Linux, do not think I make it wrong 
but will try an other terminal too

muekno, 2024-01-26:
Success tried another terminal everything is fine
no duplicate lines