banksy-git/lidl-gateway-freedom #25: Moes Tuya ZigBee Gateway Support

dmitriy-sqrt, 2022-08-22:
Hi, just bought the hub for experiments and was passing by this repo to see if instructions here will still apply for the hub i have.

So i have this version of a gateway:
[image]

Examples of aliexpress link: 
https://www.aliexpress.com/item/4000452898540.html
https://www.aliexpress.com/item/1005003190609659.html

This is how the mobo looks like:
[image]
(its pretty different, you can note it uses RTL8197FS)

And I've started my experiments.

1. Connected the UART FTDI232 to debug pins (ones near the micro usb connector), just RX\TX\GND + power via micro usb. (Although i tried to do  3.3v power only via debug port and it still seemed to work)
2. Took a while to get working serial port connection software, in the end  I used Putty on windows.
3. The loading log is a bit different from the one for lidl gateway, full gist is available [here](https://gist.github.com/dmitriy-sqrt/32195ffe2954d9a5f2733b6d7856f33c)
4. Important here that partitions (with sizes are different):
5.
6. somehow in the end i was able to get to **bootloader** with hitting `ESC` on the early load steps ()
7.

MattWestb, 2022-08-22:
Its looks very nice and 2 EFR32MG21 modules for BT and one for Zigbee so cant loading one RCP firmware on the Zigbee module and you can running BT + Zigbee and Open Thread at the same time over Ethernet :-))

Hope you can getting the root password and looking little more in the Linux tuya is using and how the hardware is configured.

MattWestb, 2022-08-22:
The SOC have 3 comports ttys0 = consolle and ttys1 and 2 plus one ch341 that is likely used for Zigbee module or for one extra USB console.

dmitriy-sqrt, 2022-08-24:
Sadly for now i cant unpack\unsquashfs dumped partitions. I've tried using `unsquashfs version 4.5.1` but get an error for:
```
Reading a different endian SQUASHFS filesystem on rootfs_v2.bin
FATAL ERROR: Can't find a valid SQUASHFS superblock on rootfs_v2.bin
```

So its either me dumping in a wrong way, or different unpacking way should be used. I'll attach the dump, in case anyone more experienced wishes to take a look
[rootfs_v2.bin.zip](https://github.com/banksy-git/lidl-gateway-freedom/files/9414834/rootfs_v2.bin.zip)

zalatnaicsongor, 2024-03-13:
the file had endianness issues, I've fixed it, it can now be extracted. @dmitriy-sqrt 
password for the 7z file within the zip:
 `dmitriy-sqrt` 
 
 [rootfs_v2_le_2.zip](https://github.com/banksy-git/lidl-gateway-freedom/files/14590873/rootfs_v2_le_2.zip)