banksy-git/lidl-gateway-freedom #27: Freed gateway disturbs Tuya sensors?

akloeckner, 2022-11-11:
Hi there,

great project, thanks!

I am having problems with my recently acquired Moes sensors: https://www.aliexpress.com/item/1005002535901726.html . It seems to be a Tuya device. I have six of them and they all behave the same.

As it appears, the sensor works well on its own if I reset it. I also coupled the sensor to a not-hacked Lidl gateway, and so far no problems.

However, as soon as I couple it to the hacked gateway (which is connected to Home Assistant through ZHA), the sensor starts reporting zeros and eventually goes into some kind of reboot loop. Here's a more detailled sequence of events:

1. Temperature and humidity sensors report Zero-Readings regularly in ZigBee, which are usually corrected after 1-3 seconds to the correct value.
2. The sensors start reporting constantly zero temperature and zero humidity at some point. (Both on the device and in ZigBee.) I have rebooted them a few times by removing the battery, but it did not help.
3. The brightness sensor reports "0" on the device and "1" in Zigbee most of the time, even if I have enough light to read in the room. The readings only go up and become conistent on the display and in ZigBee when I point a light directly on the device.
4. The display intermittently switches off (no black symbols seen) and on again. This might be related to the readings going to zero.
5. At some point the display switching happens a few times per second. And after some more time, the sensor just appears off.

Have you ever experienced a similar problem with this or a different sensor? And can you maybe propose a fix? I really like the sensors but I really do **not** want to route all my temperature readings through the Tuya cloud...

Thanks for any hint!

MattWestb, 2022-11-11:
For support of new device in ZHA pleas use https://github.com/zigpy/zha-device-handlers.
First search for your device if some one have open one issue for your model or one similar and use it i for getting it supported or if not open one new device request.

 I know your device is one of the most "exentric" tuya device and have very much tuya special that must being work around for getting it working OK but its 110% possible with brave users help.

akloeckner, 2022-11-11:
Thanks for the pointer. It appears, there is an issue already and it's closed as unfixable... Sad to read that, because the device looks great and packs a lot of features. But... There's no use if it's not usable in my setup. :-/

https://github.com/zigpy/zha-device-handlers/issues/961

I'll take some days to decide whether I will return the device or whether I will try and dig deep into the device handlers. But looking at the issue (and its long discussion) I feel a little bit desperate. ;-)

I'll close the issue here. Thanks.

closed 2022-11-11 by akloeckner

MattWestb, 2022-11-11:
You can asking reopening the issue then one dev is working very  much with strange tuya devices then hi is having time or only posting more in the closed issue.

I was having one of this on my which list but Z2M have getting it supported and its not working well at all so i have ditching it and need one light sensor or more.