banksy-git/lidl-gateway-freedom #31: Support for Tuya ZW05B0

ant-thomas, 2022-12-30:
I've picked up a cheap ethernet gateway off Aliexpress.

Seems to be the same hardware as the Lidl

Firmware is a lot newer - I've added headers and got to the boot messages so far.

Pressing ESC doesn't seem to get to the bootload prompt.

Any suggestions for next steps?

[image]
[image]
[image]

``` 
Booting...

@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
@
@ chip__no chip__id mfr___id dev___id cap___id size_sft dev_size chipSize
@ 0000000h 0c84018h 00000c8h 0000040h 0000018h 0000000h 0000018h 1000000h
@ blk_size blk__cnt sec_size sec__cnt pageSize page_cnt chip_clk chipName
@ 0010000h 0000100h 0001000h 0001000h 0000100h 0000010h 000004eh GD25Q128
@
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
DDR1:32MB

---RealTek(RTL8196E)at 2022.09.02-15:49+0800 v3.4T-pre2 [16bit](380MHz)
P0phymode=01, embedded phy
check_image_header  return_addr:05010000 bank_offset:00000000
no sys signature at 00010000!
get uboot flag failed
Jump to image start=0x80c00000...
decompressing kernel:
Uncompressing Linux... done, booting the kernel.
done decompressing kernel.
start address: 0x80003780
Linux version 3.10.90 (huangxh@embed) (gcc version 4.6.4 (Realtek RSDK-4.6.4 Build 2080) ) #5 Fri Sep 2 15:52:57 CST 2022
CPU revision is: 0000cd01
Determined physical RAM map:
 memory: 02000000 @ 00000000 (usable)
Zone ranges:
  Normal   [mem 0x00000000-0x01ffffff]
Movable zone start for each node
Early memory node ranges
  node   0: [mem 0x00000000-0x01ffffff]
icache: 16kB/16B, dcache: 8kB/16B, scache: 0kB/0B
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 8128
Kernel command line:  console=ttyS0,38400 root=/dev/mtdblock2
PID hash table entries: 128 (order: -3, 512 bytes)
Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
Inode-cache hash table entries: 2048 (order: 1, 8192 bytes)
Memory: 27344k/32768k available (2763k kernel code, 5424k reserved, 562k data, 192k init, 0k highmem)
SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
NR_IRQS:128
console [ttyS0] enabled
Calibrating delay loop... 378.47 BogoMIPS (lpj=1892352)
pid_max: default: 4096 minimum: 301
Mount-cache hash table entries: 512
reg e0=0
reg e1=0
reg e2=0
reg e3=0
reg e4=0
reg e5=0
reg e6=0
reg e7=0
reg f0=0
reg f1=0
reg f2=0
reg f3=0
reg f4=0
reg f5=0
reg f6=0
NET: Registered protocol family 16
bio: create slab <bio-0> at 0
NET: Registered protocol family 2
TCP established hash table entries: 512 (order: 0, 4096 bytes)
TCP bind hash table entries: 512 (order: -1, 2048 bytes)
TCP: Hash tables configured (established 512 bind 512)
TCP: reno registered
UDP hash table entries: 256 (order: 0, 4096 bytes)
UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
NET: Registered protocol family 1
squashfs: version 4.0 (2009/01/31) Phillip Lougher
jffs2: version 2.2. (NAND) © 2001-2006 Red Hat, Inc.
msgmni has been set to 53
Block layer SCSI generic (bsg) driver version 0.4 loaded (major 254)
io scheduler noop registered
io scheduler deadline registered
io scheduler cfq registered (default)
Serial: 8250/16550 driver, 2 ports, IRQ sharing disabled
serial8250: ttyS0 at MMIO 0x18002000 (irq = 9) is a 16550A
serial8250: ttyS1 at MMIO 0x18002100 (irq = 13) is a 16550A
Realtek GPIO Driver for Flash Reload Default
tuya_gpio_init ok, scan expire time:50
SPI INIT
 ------------------------- Force into Single IO Mode ------------------------
|No chipID  Sft chipSize blkSize secSize pageSize sdCk opCk      chipName    |
| 0 c84018h  0h 1000000h  10000h  10000h     100h   84    0          GD25Q128|
 ----------------------------------------------------------------------------
SPI flash(GD25Q128) was found at CS0, size 0x1000000
boot+cfg offset=0x0 size=0x20000 erasesize=0x10000
linux offset=0x20000 size=0x1e0000 erasesize=0x10000
rootfs offset=0x200000 size=0x200000 erasesize=0x10000
tuya-label offset=0x400000 size=0x20000 erasesize=0x10000
jffs2-fs offset=0x420000 size=0xbe0000 erasesize=0x10000
5 rtkxxpart partitions found on MTD device flash_bank_1
Creating 5 MTD partitions on "flash_bank_1":
0x000000000000-0x000000020000 : "boot+cfg"
0x000000020000-0x000000200000 : "linux"
0x000000200000-0x000000400000 : "rootfs"
0x000000400000-0x000000420000 : "tuya-label"
0x000000420000-0x000001000000 : "jffs2-fs"
PPP generic driver version 2.4.2
nf_conntrack version 0.5.0 (427 buckets, 1708 max)
ip_tables: (C) 2000-2006 Netfilter Core Team
TCP: cubic registered
NET: Registered protocol family 10
sit: IPv6 over IPv4 tunneling driver
NET: Registered protocol family 17
l2tp_core: L2TP core driver, V2.0
8021q: 802.1Q VLAN Support v1.8
Realtek FastPath:v1.03

Probing RTL819X NIC-kenel stack size order[1]...
eth0 added. vid=9 Member port 0x10f...
eth1 added. vid=8 Member port 0x10...
[peth0] added, mapping to [eth1]...
VFS: Mounted root (squashfs filesystem) readonly on device 31:2.
Freeing unused kernel memory: 192K (80340000 - 80370000)
init started: BusyBox v1.13.4 (2022-09-02 15:48:01 CST)
Set power startcmd read

b8000038: 2794A104  0000000F    00000042  00000018    '▒▒        B
cmd write
Write memory 0xb8000038 dat 0x1794a104: 0x1794a104
Set power end
killall: dropbear: no process killed

Please press Enter to activate this console. udhcpc (v1.13.4) started
Tuya Gateway Application Normal Srart /tuya/tuya_start.sh UserAppRunDir:
set defult run_dir:/tuya
TY_ENV_APP_RUN_DIR=/tuya
Sending discover...
get user cfg file error, load defult cfg file
load platform configure file:/tuya/def.cfg
start.conf is exist
udhcpc (v1.13.4) started
current run dir:/tuya/tuya_user1
tuya_start_children.sh:UserAppRunDir:/tuya JsonFile Path:/tuya/def.cfg [engineer_mode: ]
grep: /var/resolv.conf: No such file or directory
Sending discover...
killall: app_detect.sh: no process killed
killall: tyZ3Gw: no process killed
killall: log_detect.sh: no process killed
killall: process_monitor.sh: no process killed
killall: tyZ3Gw: no process killed
killall: dropbear: no process killed
Sending discover...
cat: can't open '/tuya/eng_mode': No such file or directory
no eng file
Sending discover...
nlRecvFromAppSock sg_netlinkKeyPid:242
nlRecvFromAppSock port link sg_netlinkPid:242
nameserver 8.8.8.8
nameserver 114.114.114.114
Sending discover...
Sending discover...
Sending discover...
```

MattWestb, 2022-12-30:
Is you sure that the TX is working OK from your TTL adapter ??
Also make sure both hard and software flow control is disable of the TTL adapter is not sending any commands to the device. The baud rate and bits is OK then you is getting OK boot log in the terminal.

Its looks very nice and is using the same Zigbee module but the IPX version and i hope they have not changing the pins / pads used on it so we can using  the same firmware as the original one.

ant-thomas, 2022-12-30:
TX is definitely working as I'm able to press enter to get to console login

```
Please press Enter to activate this console.
```

I'll have a look at the flow control and play some more.

MattWestb, 2022-12-30:
That good then TX is working in the "cable".
Its reported not so easy getting in the boolader in some newer versions of the first gen ZBGW so you is not alone.

MattWestb, 2022-12-30:
I was finding one tuya ZBGW that looks similar but little different and its possible hacking :-)))

https://kaspars.net/blog/zemismart-zigbee-homekit-hub

ant-thomas, 2022-12-30:
I haven't connected it to the Tuya cloud app yet.
Should I set that up then try to hack?

ant-thomas, 2022-12-30:
I've added it to the cloud app.
Looks like the SSH server isn't running - or at least not on port 2333 and there is mention of dropbear being killed during the original boot output.

Open ports
6668
6682 (when I try to ssh - kex_exchange_identification: Connection closed by remote host - so maybe ssh server on this?)
12130

fcollingwood, 2023-03-02:
Exact same problem with the exact same hardware.

Keystrokes are disabled during the boot process until Busybox is loaded, thus it is impossible to to stop the bootloader

fcollingwood, 2023-03-02:
> I was finding one tuya ZBGW that looks similar but little different and its possible hacking :-)))
> 
> https://kaspars.net/blog/zemismart-zigbee-homekit-hub

Irrelevant. Requires being able to break out of the bootloader, which is impossible to do

fcollingwood, 2023-03-21:
I have tried three different hubs now, and can not break out during the boot process. I have tried holding down the ESC key while powering them, have tried constantly hitting ESC while powering on, but they just do not break out. It seems that the current firmware version disables reading keystrokes until the boot cycle is almost complete.

Does anyone have a solution for this issue?

Tonybarrett62, 2023-07-27:
I've been playing with this device and found that if you short the pins 6-7 together shortly after powering on the device with a microusb cable.
You need to allow it to read the bootloader from the memory and then bridge the pins when it attempt to read the rootfs partition, approx 1 sec after power on.

[image]
[image]

Booting...                                                                      
                                                                                
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
@                                                                               
@ chip__no chip__id mfr___id dev___id cap___id size_sft dev_size chipSize       
@ 0000000h 0c84018h 00000c8h 0000040h 0000018h 0000000h 0000018h 1000000h       
@ blk_size blk__cnt sec_size sec__cnt pageSize page_cnt chip_clk chipName       
@ 0010000h 0000100h 0001000h 0001000h 0000100h 0000010h 000004eh GD25Q128       
@                                                                               
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
DDR1:32MB                                                                       
                                                                                
---RealTek(RTL8196E)at 2022.09.02-15:49+0800 v3.4T-pre2 [16bit](400MHz)         
P0phymode=01, embedded phy                                                      
check_image_header  return_addr:05010000 bank_offset:00000000                   
no sys signature at 00010000!                                                   
rootfs checksum error at 00200000!                                              
no rootfs signature at 00210000!                                                
no rootfs signature at 00250000!                                                
no rootfs signature at 00201000!                                                
no rootfs signature at 00202000!                                                
no rootfs signature at 00203000!                                                
no rootfs signature at 00204000!                                                
no rootfs signature at 00205000!                                                
no rootfs signature at 00206000!                                                
no rootfs signature at 00207000!                                                
no rootfs signature at 00208000!                                                
no rootfs signature at 00209000!      
.......................................................................................
no rootfs signature at 003F6000!                                                
no rootfs signature at 003F7000!                                                
no rootfs signature at 003F8000!                                                
no rootfs signature at 003F9000!                                                
no rootfs signature at 003FA000!                                                
no rootfs signature at 003FB000!                                                
no rootfs signature at 003FC000!                                                
no rootfs signature at 003FD000!                                                
no rootfs signature at 003FE000!                                                
no rootfs signature at 003FF000!                                                
no rootfs signature at 00400000!                                                
get uboot flag failed                                                           
P0phymode=01, embedded phy                                                      
                                                                                
---Ethernet init Okay!                                                          
<RealTek>

Dont know if this helps, 
In the end i needed to buy a chip reader and copy the chips memory, replace the password file and write it back to the chip. I was then able to copy serialgateway.sh and enable the tuya ssh server. Currently connected to my HA using ZHA and some tuya motion sensors.

G414ng, 2023-11-21:
I found the way

firuz1844, 2024-05-07:
> I found the way

Tell us more plz)

robsonfelix, 2024-11-18:
I got to the same place @Tonybarrett62 did, but could not go further. @G414ng please tell us what you did....

newicody, 2025-01-20:
ty