banksy-git/lidl-gateway-freedom #33: Revision 2 of the gateway

Dexruus, 2023-02-10:
Less of an Issue and more of an "advice"; Revision 2 of the Lidl gateway (rounded edges) is the "exact same". Its firmware just has ssh disabled. You need to TTL onto the device and start its dropbear ssh server with;

dropbear -p "port"

From thereon you are able to ssh into the device with your given root until next restart.

Its recommended to follow the tutorial part in which you set the ssh port to "always on" and "port 22":

if [ ! -f /tuya/ssh_monitor.original.sh ]; then cp /tuya/ssh_monitor.sh /tuya/ssh_monitor.original.sh; fi

echo "#!/bin/sh" >/tuya/ssh_monitor.sh

ALSO FOR NOOBS LIKE ME;

IF:

Unable to negotiate with "ip.address" "port" : no matching host key type found. Their offer: ssh-rsa,ssh-dss

USE:

ssh -o HostKeyAlgorithms=+ssh-rsa "ip-address" -p "port"

bmix, 2024-03-20:
Would you mind uploading a photo of those parts of the device where one could draw conclusions as to how to open that darn thing?  How did you get it to open?
Thanks!

GAS85, 2024-04-01:
> Would you mind uploading a photo of those parts of the device where one could draw conclusions as to how to open that darn thing? How did you get it to open? Thanks!
> 
> I have the rev 2, but all instructions help is for rev 1.

Have had the same issue. Problem was in USB connection of USB TTY device. Pull it out and in - problem was solved.

GAS85, 2024-04-01:
> Its firmware just has ssh disabled.

Not really, it is simply running on port `2333`, this is not standard port of ssh. So, to connect you have to run command (there is no space in `-p2333`):

```shell
ssh -o HostKeyAlgorithms=+ssh-rsa root@<ip-address> -p2333
```