banksy-git/lidl-gateway-freedom #37: Lidl Silvercrest Zigbee Gateway Model  ZG071320

TheRealRWJ, 2023-10-12:
First of all: Thanks  for this project

Is this project still valid for the newer model of the Lidl Silvercrest Zigbee Gateway Model  ZG071320.
I followed your steps on https://paulbanks.org/projects/lidl-zigbee/

I do not get any output in Minicom

Thanks for any tips

xdubx, 2024-02-26:
The product number is SGWZ 1 A2?

xdubx, 2024-02-26:
Did you try the extra step from https://github.com/banksy-git/lidl-gateway-freedom/issues/33 ?

xdubx, 2024-02-26:
Oh sry its late. Witch ttl adapter did you use? Or you directly wired to an rs232 port at your computer?

TheRealRWJ, 2024-02-26:
Late? Only in thuis part of the world. No problemen.

I used a pl2303 RTL USB serial rs232 from otronic

I saw some data once while booting.
Never saw that again.
I assumed the gateway got updated and the option got 'fixed'

a3rosol, 2024-03-10:
So I have the same gateway (HG07132) and when I connect to it using tio/screen (mac os sonoma if that matters) I receive a bunch of data but none of it is human readable. 

`tio /dev/tty.usbserial-B002YYQH -b 38400 -d 8 -p none -s 1 --flow none` is the command I use to connect to it. I receive a bunch of data and pressing ESC doesn't seem to help tbh. I also used a different USB TTL board with a ch340g instead of a FT232RL I get the same result sadly. Any pointers would be really appreciated

GAS85, 2024-04-01:
Actually works for me fine if set correct bits:
```
sudo screen /dev/ttyUSB0 38400
```

GAS85, 2024-04-01:
> So I have the same gateway (HG07132) and when I connect to it using tio/screen (mac os sonoma if that matters) I receive a bunch of data but none of it is human readable.

Have had the same issue. Problem was in USB connection of USB TTY device. Pull it out and in - problem was solved.

sbonaime, 2026-03-13:
> So I have the same gateway (HG07132) and when I connect to it using tio/screen (mac os sonoma if that matters) I receive a bunch of data but none of it is human readable.
>
> `tio [...]

I have the same problem :-( Any idea ? Thanks