banksy-git/lidl-gateway-freedom #42: lidl_auskey_decode.py do not run

raveit65, 2024-07-11:
The script always give me the error message.
```
[rave@satellite Downloads]$ python3 lidl_auskey_decode-new.py 
Enter KEK hex string line>80000000:	FFFFFFFF	FFFFFFFF	FFFFFFFFFFFFFFFF
Encoded aus-key as hex string line 1>80000000:	FFFFFFFF	FFFFFFFF	FFFFFFFF	FFFFFFFF
Encoded aus-key as hex string line 2>80000010:	FFFFFFFF	FFFFFFFF	FFFFFFFF	FFFFFFFF
Traceback (most recent call last):
  File "/home/rave/Downloads/lidl_auskey_decode-new.py", line 65, in <module>
    print("Auskey:", auskey.decode("ascii"))
                     ^^^^^^^^^^^^^^^^^^^^^^
UnicodeDecodeError: 'ascii' codec can't decode byte 0xec in position 0: ordinal not in range(128)
[rave@satellite Downloads]$
```
System is fedora 40 with `python3-3.12.3-2.fc40.x86_64`

I am also wondering about the output of kek and auskey in serial console.
```
<RealTek>FLR 80000000 401802 16
Flash read from 00401802 to 80000000 with 00000016 bytes	?
(Y)es , (N)o ? --> y
Flash Read Successed!
<RealTek>DW 80000000 4
80000000:	FFFFFFFF	FFFFFFFF	FFFFFFFF	FFFFFFFF
<RealTek>
<RealTek>FLR 80000000 402002 32
Flash read from 00402002 to 80000000 with 00000032 bytes	?
(Y)es , (N)o ? --> y
Flash Read Successed!
<RealTek>DW 80000000 8
80000000:	FFFFFFFF	FFFFFFFF	FFFFFFFF	FFFFFFFF
80000010:	FFFFFFFF	FFFFFFFF	FFFFFFFF	FFFFFFFF
<RealTek>
```
I post this in public because i think the output is wrong.
Can you help me please to get the root password of my lidl gateway?

matze20211, 2024-12-30:
Here too:

`<RealTek>FLR 80000000 401802 16                                                 
Flash read from 00401802 to 80000000 with 00000016 bytes        ?               
(Y)es , (N)o ? --> Y                                                            
Flash Read Successed!                                                           
<RealTek>DW 80000000 4                                                          
80000000:       FFFFFFFF        FFFFFFFF        FFFFFFFF        FFFFFFFF        
<RealTek>FLR 80000000 402002 32                                                 
Flash read from 00402002 to 80000000 with 00000032 bytes        ?               
(Y)es , (N)o ? --> Y                                                            
Flash Read Successed!                                                           
<RealTek>DW 80000000 8                                                          
80000000:       FFFFFFFF        FFFFFFFF        FFFFFFFF        FFFFFFFF        
80000010:       FFFFFFFF        FFFFFFFF        FFFFFFFF        FFFFFFFF `

searching for a solution i found: some using other adresses, but i dont know whre to find, here in?

` ------------------------- Force into Single IO Mode ------------------------   
|No chipID  Sft chipSize blkSize secSize pageSize sdCk opCk      chipName    |  
| 0 c84018h  0h 1000000h  10000h  10000h     100h   84    0          GD25Q128|  
 ----------------------------------------------------------------------------   
SPI flash(GD25Q128) was found at CS0, size 0x1000000                            
boot+cfg offset=0x0 size=0x20000 erasesize=0x10000                              
linux offset=0x20000 size=0x1e0000 erasesize=0x10000                            
rootfs offset=0x200000 size=0x200000 erasesize=0x10000                          
tuya-label offset=0x400000 size=0x20000 erasesize=0x10000                       
jffs2-fs offset=0x420000 size=0xbe0000 erasesize=0x10000                        
5 rtkxxpart partitions found on MTD device flash_bank_1                         
Creating 5 MTD partitions on "flash_bank_1":                                    
0x000000000000-0x000000020000 : "boot+cfg"                                      
0x000000020000-0x000000200000 : "linux"                                         
0x000000200000-0x000000400000 : "rootfs"                                        
0x000000400000-0x000000420000 : "tuya-label"                                    
0x000000420000-0x000001000000 : "jffs2-fs"             `

matze

0xD4M13N, 2026-01-17:
> The script always give me the error message.
>
> ```
> [rave@satellite Downloads]$ python3 lidl_auskey_decode-new.py
> Enter KEK hex string line>80000000:	FFFFFFFF	FFFFFFFF	FFFFFFFFFFFFFFFF
> Encoded [...]

It's maybe a little bit too late, but if it can help you (or somebody else), your problem is that you send each line with its address.
Your input shouldn't look like :
80000000:	FFFFFFFF	FFFFFFFF	FFFFFFFFFFFFFFFF
but :
FFFFFFFF	FFFFFFFF	FFFFFFFFFFFFFFFF