banksy-git/lidl-gateway-freedom #43: Failed setup, will retry: <XncpCommandId.undefined_0x0100: 256>

robsonfelix, 2024-12-09:
mine was working until the update last week. After updating Home Assistant to version 2024.12.1 it just stopped working. it gives me error “Failed setup, will retry: <XncpCommandId.undefined_0x0100: 256>”. Anyone with the same issue?

here is the debug log:

2024-12-09 08:44:07.684 DEBUG (MainThread) [zigpy.serial] Opening a serial connection to ‘socket://10.0.1.202:8888’ (baudrate=115200, xonxoff=True, rtscts=False)
2024-12-09 08:44:07.687 DEBUG (MainThread) [zigpy.serial] Connection made: <bellows.ash.AshProtocol object at 0x7f4c43cb3070>
2024-12-09 08:44:08.688 DEBUG (MainThread) [bellows.ezsp] Resetting EZSP
2024-12-09 08:44:08.688 DEBUG (MainThread) [bellows.uart] Resetting ASH
2024-12-09 08:44:08.688 DEBUG (MainThread) [bellows.ash] Sending frame CANCEL + RstFrame() + FLAG
2024-12-09 08:44:08.688 DEBUG (MainThread) [bellows.ash] Sending data 1ac038bc7e
2024-12-09 08:44:09.085 WARNING (MainThread) [homeassistant.components.sensor] Updating solarman sensor took longer than the scheduled update interval 0:00:30
2024-12-09 08:44:09.120 WARNING (SyncWorker_4) [custom_components.solarman.solarman] Querying [1 - 125] failed with exception [Empty: ]
2024-12-09 08:44:09.120 WARNING (SyncWorker_4) [custom_components.solarman.solarman] Querying [1 - 125] failed, [0] retry attempts left
2024-12-09 08:44:09.120 WARNING (SyncWorker_4) [custom_components.solarman.solarman] Querying registers [1 - 125] failed, aborting.
2024-12-09 08:44:09.770 DEBUG (MainThread) [bellows.ash] Received data 1ac1020b0a527e
2024-12-09 08:44:09.770 DEBUG (MainThread) [bellows.ash] Received cancel byte, clearing buffer
2024-12-09 08:44:09.770 DEBUG (MainThread) [bellows.ash] Received frame RStackFrame(version=2, reset_code=<NcpResetCode.RESET_SOFTWARE: 11>)
2024-12-09 08:44:09.772 DEBUG (MainThread) [bellows.ezsp] Switching to EZSP protocol version 4
2024-12-09 08:44:09.772 DEBUG (MainThread) [bellows.ezsp.protocol] Sending command version: () {‘desiredProtocolVersion’: 4}
2024-12-09 08:44:09.772 DEBUG (MainThread) [bellows.ash] Sending frame DataFrame(frm_num=0, re_tx=False, ack_num=0, ezsp_frame=b’\x00\x00\x00\x04’) + FLAG
2024-12-09 08:44:09.772 DEBUG (MainThread) [bellows.ash] Sending data 004221a850ed2c7e
2024-12-09 08:44:09.776 DEBUG (MainThread) [bellows.ash] Received data 0142a1a8532845d7
2024-12-09 08:44:09.818 DEBUG (MainThread) [bellows.ash] Received data cf007e
2024-12-09 08:44:09.819 DEBUG (MainThread) [bellows.ash] Received frame DataFrame(frm_num=0, re_tx=0, ack_num=1, ezsp_frame=b’\x00\x80\x00\x07\x02Pe’)
2024-12-09 08:44:09.819 DEBUG (MainThread) [bellows.ash] Sending frame AckFrame(res=0, ncp_ready=0, ack_num=1) + FLAG
2024-12-09 08:44:09.819 DEBUG (MainThread) [bellows.ash] Sending data 8160597e
2024-12-09 08:44:09.819 DEBUG (MainThread) [bellows.ezsp.protocol] Received command version: {‘protocolVersion’: 7, ‘stackType’: 2, ‘stackVersion’: 25936}
2024-12-09 08:44:09.819 DEBUG (MainThread) [bellows.ash] Changing ACK timeout from 1.60 to 1.42
2024-12-09 08:44:09.819 DEBUG (MainThread) [bellows.ezsp] Switching to EZSP protocol version 7
2024-12-09 08:44:09.820 DEBUG (MainThread) [bellows.ezsp.protocol] Sending command version: () {‘desiredProtocolVersion’: 7}
2024-12-09 08:44:09.820 DEBUG (MainThread) [bellows.ash] Sending frame DataFrame(frm_num=1, re_tx=False, ack_num=1, ezsp_frame=b’\x00\x00\xff\x00\x00\x07’) + FLAG
2024-12-09 08:44:09.820 DEBUG (MainThread) [bellows.ash] Sending data 7d31422157542a1240277e
2024-12-09 08:44:09.825 DEBUG (MainThread) [bellows.ash] Received data 1242a157542a12b0
2024-12-09 08:44:09.866 DEBUG (MainThread) [bellows.ash] Received data 09f175cb7e
2024-12-09 08:44:09.867 DEBUG (MainThread) [bellows.ash] Received frame DataFrame(frm_num=1, re_tx=0, ack_num=2, ezsp_frame=b’\x00\x80\xff\x00\x00\x07\x02Pe’)
2024-12-09 08:44:09.867 DEBUG (MainThread) [bellows.ash] Sending frame AckFrame(res=0, ncp_ready=0, ack_num=2) + FLAG
2024-12-09 08:44:09.867 DEBUG (MainThread) [bellows.ash] Sending data 82503a7e
2024-12-09 08:44:09.867 DEBUG (MainThread) [bellows.ezsp.protocol] Received command version: {‘protocolVersion’: 7, ‘stackType’: 2, ‘stackVersion’: 25936}
2024-12-09 08:44:09.867 DEBUG (MainThread) [bellows.ash] Changing ACK timeout from 1.42 to 1.27
2024-12-09 08:44:09.867 DEBUG (MainThread) [bellows.ezsp] EZSP Stack Type: 2, Stack Version: 6550, Protocol version: 7
2024-12-09 08:44:09.867 DEBUG (MainThread) [bellows.ezsp] Sending XNCP frame: XncpCommand(command_id=<XncpCommandId.GET_SUPPORTED_FEATURES_REQ: 0>, status=<EmberStatus.SUCCESS: 0>, payload=GetSupportedFeaturesReq())
2024-12-09 08:44:09.868 DEBUG (MainThread) [bellows.ezsp.protocol] Sending command customFrame: (b’\x00\x00\x00’,) {}
2024-12-09 08:44:09.868 DEBUG (MainThread) [bellows.ash] Sending frame DataFrame(frm_num=2, re_tx=False, ack_num=2, ezsp_frame=b’\x01\x00\xff\x00G\x03\x00\x00\x00’) + FLAG
2024-12-09 08:44:09.868 DEBUG (MainThread) [bellows.ash] Sending data 22432157546d16b259942e637e
2024-12-09 08:44:09.873 DEBUG (MainThread) [bellows.ash] Received data 2343a157546d15b1
2024-12-09 08:44:09.915 DEBUG (MainThread) [bellows.ash] Received data 5995495e857e
2024-12-09 08:44:09.915 DEBUG (MainThread) [bellows.ash] Received frame DataFrame(frm_num=2, re_tx=0, ack_num=3, ezsp_frame=b’\x01\x80\xff\x00G\x00\x03\x00\x01\x03’)
2024-12-09 08:44:09.915 DEBUG (MainThread) [bellows.ash] Sending frame AckFrame(res=0, ncp_ready=0, ack_num=3) + FLAG
2024-12-09 08:44:09.915 DEBUG (MainThread) [bellows.ash] Sending data 83401b7e
2024-12-09 08:44:09.915 DEBUG (MainThread) [bellows.ezsp.protocol] Received command customFrame: {‘status’: <EmberStatus.SUCCESS: 0>, ‘reply’: b’\x00\x01\x03’}
2024-12-09 08:44:09.915 DEBUG (MainThread) [bellows.ash] Changing ACK timeout from 1.27 to 1.13
2024-12-09 08:44:09.915 DEBUG (MainThread) [zigpy.serial] Waiting for serial port to close
2024-12-09 08:44:09.915 DEBUG (MainThread) [zigpy.serial] Connection lost: None
2024-12-09 08:44:09.915 DEBUG (MainThread) [bellows.uart] Connection lost: None
2024-12-09 08:44:09.915 DEBUG (MainThread) [zigpy.application] Connection to the radio has been lost: None
2024-12-09 08:44:09.915 DEBUG (MainThread) [zha.application.helpers] stopping global updater
2024-12-09 08:44:09.915 DEBUG (MainThread) [zha.application.helpers] global updater stopped
2024-12-09 08:44:09.915 DEBUG (MainThread) [zha.application.helpers] stopping device availability checker
2024-12-09 08:44:09.915 DEBUG (MainThread) [zha.application.helpers] device availability checker stopped
2024-12-09 08:44:09.916 DEBUG (MainThread) [zha.application.gateway] Shutting down ZHA ControllerApplication
2024-12-09 08:44:09.916 DEBUG (Thread-42146) [aiosqlite] executing functools.partial(<function PersistingListener._set_isolation_level.. at 0x7f4c91968180>)
2024-12-09 08:44:09.916 DEBUG (Thread-42146) [aiosqlite] operation functools.partial(<function PersistingListener._set_isolation_level.. at 0x7f4c91968180>) completed
2024-12-09 08:44:09.916 DEBUG (Thread-42146) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f4c5e4fe2f0>, ‘PRAGMA wal_checkpoint;’, )
2024-12-09 08:44:09.916 DEBUG (Thread-42146) [aiosqlite] operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f4c5e4fe2f0>, ‘PRAGMA wal_checkpoint;’, ) completed
2024-12-09 08:44:09.917 DEBUG (Thread-42146) [aiosqlite] executing functools.partial(<function PersistingListener._set_isolation_level.. at 0x7f4c5c405800>)
2024-12-09 08:44:09.917 DEBUG (Thread-42146) [aiosqlite] operation functools.partial(<function PersistingListener._set_isolation_level.. at 0x7f4c5c405800>) completed
2024-12-09 08:44:09.917 DEBUG (Thread-42146) [aiosqlite] executing functools.partial(<built-in method close of sqlite3.Connection object at 0x7f4c5e4fe2f0>)
2024-12-09 08:44:09.917 DEBUG (Thread-42146) [aiosqlite] operation functools.partial(<built-in method close of sqlite3.Connection object at 0x7f4c5e4fe2f0>) completed
2024-12-09 08:44:10.021 DEBUG (MainThread) [homeassistant.components.zha] Failed to set up ZHA
Traceback (most recent call last):
[... 1 lines cut ...]
await zha_gateway.async_initialize()
File “/usr/local/lib/python3.13/site-packages/zha/application/gateway.py”, line 275, in async_initialize
await self._async_initialize()
File “/usr/local/lib/python3.13/site-packages/zha/application/gateway.py”, line 258, in _async_initialize
await self.application_controller.startup(auto_form=True)
[... cut ...]
File “/usr/local/lib/python3.13/site-packages/bellows/ezsp/init.py”, line 679, in xncp_get_supported_firmware_features
rsp = await self.send_xncp_frame(xncp.GetSupportedFeaturesReq())
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File “/usr/local/lib/python3.13/site-packages/bellows/ezsp/init.py”, line 663, in send_xncp_frame
rsp_frame = xncp.XncpCommand.from_bytes(data)
File “/usr/local/lib/python3.13/site-packages/bellows/ezsp/xncp.py”, line 70, in from_bytes
payload, rest = COMMANDS[command_id].deserialize(data)
~~~~~~~~^^^^^^^^^^^^
KeyError: <XncpCommandId.undefined_0x0100: 256>

vanupied, 2024-12-09:
Hello,
Same issue here. Since last home assistant update to  2024.12.1, my already configured and working lidl gateway suddenly stopped working.
I tried to remove it in order to declare it again, but it appear home assistant is unable to communicate with the gateway anymore.
You're not alone @robsonfelix :)

mdaitn, 2024-12-13:
Hey, you need to update the firmware your gateway:
https://community.home-assistant.io/t/zha-ezsp-gateway-lidl-cant-initialize-after-update-to-ha-24-12-0-xncpcommandid-undefined-0x03e8-1000/805164/5

vanupied, 2025-01-07:
Thanks for the hint ! issue can be closed as the firmware update solved the issue